usdot-fhwa-stol/carma-platform #112: LocalSpeedLimitConstraint fails to find local speed at the end of the route

qswawrq, 2019-01-10:
In 10/17 Glidepath testing, we encounter the following Exception:
``` 2018-10-17 16:26:58:878 | INFO | CruisingPlugin | NONE | Planned maneuver from [11.18, 11.18) m/s over [416.10, 642.94) m 
 2018-10-17 16:26:58:878 | WARNING | LocalSpeedLimitConstraint | NONE | Maneuver from [416.10, 642.94) deemed illegal due to missing speed limits 
 2018-10-17 16:26:58:878 | WARNING | TrajectoryValidator | NONE | Trajectory from [416.10, 642.94) failed validation on constraint: LocalSpeedLimitConstraint for reason: Maneuvers exceed route defined Speed Limit! 
 2018-10-17 16:26:58:878 | WARNING | TrajectoryValidator | NONE | Offending maneuvers: {SteadySpeed@[416.10, 642.94)} 
 2018-10-17 16:26:58:878 | WARNING | Arbitrator | NONE | Candidate trajectory #5 failed validation. 
 2018-10-17 16:26:58:878 | SEVERE | GuidanceExceptionHandler | NONE | Arbitrator unable to plan valid trajectory after 5 attempts! 
org.ros.exception.RosRuntimeException: Unable to plan trajectory
	at gov.dot.fhwa.saxton.carma.guidance.arbitrator.Arbitrator.planTrajectory(Arbitrator.java:425)
	at gov.dot.fhwa.saxton.carma.guidance.arbitrator.Arbitrator.failedTrajectoryReplan(Arbitrator.java:567)
	at gov.dot.fhwa.saxton.carma.guidance.GuidanceComponent.runJobQueue(GuidanceComponent.java:135)
	at gov.dot.fhwa.saxton.carma.guidance.GuidanceComponent$1.loop(GuidanceComponent.java:154)
	at org.ros.concurrent.CancellableLoop.run(CancellableLoop.java:56)
	at java.lang.Thread.run(Thread.java:748)
```
LocalSpeedLimitConstraint fails to find local speed at the end of the route and treats the last maneuver as illegal, which prevents the CruisingPlugin from planning to the end of the route.
See attached log file:
[guidance.txt](https://github.com/usdot-fhwa-stol/CARMAPlatform/files/2489174/guidance.txt)

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 1077

TonysCousin, 2020-04-17:
This applies to the Carma2 architecture only, so is now OBE.

closed 2020-04-17 by TonysCousin