usdot-fhwa-stol/carma-platform #120: Truck CAN interface configuration depends on the order of CAN device initialization

qswawrq, 2019-01-10:
Each driver needs to connect to a specific PC CAN interface. But truck CAN interface will be reordered if CAN devices initialized in a different order. For example if we unplug the radar USB PCAN, it will change the name of truck controller driver CAN interface name (from can1 to can0) and truck CAN driver (from can2 to can1).
Ideally, we should set up some udev rules on the computer to lock in the devices regardless of what order they are initialized.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 989

maefromm, 2020-01-15:
@msmcconnell @qswawrq is this still an open issue for what's on the red truck?

TonysCousin, 2020-05-05:
It seems this was an issue specific to the old CARMA2 computers, and has not been experienced in any CARMA3 installations, so closing as OBE.

closed 2020-05-05 by TonysCousin