usdot-fhwa-stol/carma-platform #122: Occasionally Arbitrator Ignores SpeedHarm Plugin

qswawrq, 2019-01-10:
During a roughly 4 min I-95 test run, Arbitrator is not asking SpeedHarm plugin to plan a new trajectory for two times. It is not clear what cause that. But it makes the vehicle be out of control of the speed harm server and back to the control of cruising.
The timestamp is 2018-06-19-10-28-44. Here is a snippet of the log:
```
 2018-06-19 10:53:37:744 | WARNING | Arbitrator | NONE | Using experimental replan method, replanning with window: 38987.46337911795 
 2018-06-19 10:53:37:744 | INFO | ArbitratorStateMachine | NONE | Arbitrator processing event: TRAJECTORY_FAILED_EXECUTION 
 2018-06-19 10:53:37:744 | INFO | ArbitratorStateMachine | NONE | Arbitrator transitioned to state REPLAN_DUE_TO_FAILED_TRAJECTORY 
 2018-06-19 10:53:37:744 | INFO | Arbitrator | NONE | Arbitrator replanning trajectory because of failed execution! 
 2018-06-19 10:53:37:745 | INFO | Arbitrator | NONE | Arbitrator planning new trajectory spanning [19190.049137347014, 30569.189513248486) 
 2018-06-19 10:53:37:745 | INFO | Arbitrator | NONE | Allowing plugin: Platooning Plugin to plan trajectory. 
 2018-06-19 10:53:37:746 | INFO | PlatooningPlugin | NONE | Plan Trajectory from Trajectory [startLocation=19190.049137347014, endLocation=30569.189513248486] in state PlatoonLeaderState 
 2018-06-19 10:53:37:746 | INFO | ConflictManager | NONE | Adding requested path 
 2018-06-19 10:53:37:746 | INFO | ConflictManager | NONE | addPath 
 2018-06-19 10:53:37:747 | INFO | ConflictManager | NONE | Preparing to insert points 
 2018-06-19 10:53:37:747 | INFO | ConflictManager | NONE | Inserting path with size 1 
 2018-06-19 10:53:37:747 | INFO | ConflictManager | NONE | Done inserting 
 2018-06-19 10:53:37:748 | INFO | Arbitrator | NONE | Allowing plugin: Route Following Plugin to plan trajectory. 
 2018-06-19 10:53:37:748 | INFO | RouteFollowingPlugin | NONE | Route Following Plugin planning trajectory! 
 2018-06-19 10:53:37:748 | INFO | RouteFollowingPlugin | NONE | Identified 0 lane changes in trajectory 
 2018-06-19 10:53:37:748 | INFO | RouteFollowingPlugin | NONE | Changes:  
 2018-06-19 10:53:37:748 | INFO | RouteFollowingPlugin | NONE | Planning lane keeping maneuver planning between: [19190.05, 30569.19) 
 2018-06-19 10:53:37:749 | INFO | RouteFollowingPlugin | NONE | Route Following planning success! 
 2018-06-19 10:53:37:749 | INFO | Arbitrator | NONE | Allowing plugin: Cruising Plugin to plan trajectory. 
 2018-06-19 10:53:37:749 | INFO | CruisingPlugin | NONE | Found 1 empty gaps in trajectory [19190.049137347014, 30569.189513248486] 
 2018-06-19 10:53:37:749 | INFO | CruisingPlugin | NONE | Gap: [startLocation = 19190.049137347014; endLocation = 30569.189513248486; startSpeed = 24.785999298095703] 
 2018-06-19 10:53:37:750 | INFO | VehicleAwareness | NONE | Notified of a forced replan, cleaning invalid trajectories! 
 2018-06-19 10:53:37:752 | INFO | CruisingPlugin | NONE | Found 1 speed limits in Gap: [startLocation = 19190.049137347014; endLocation = 30569.189513248486; startSpeed = 24.785999298095703] 
 2018-06-19 10:53:37:752 | INFO | CruisingPlugin | NONE | SpeedLimit: [location = 30569.189513248486, limit = 29.0576] 
 2018-06-19 10:53:37:752 | INFO | CruisingPlugin | NONE | Trying to plan maneuver {start=19190.05,end=30569.19,startSpeed=24.79,endSpeed=29.06} 
 2018-06-19 10:53:37:753 | INFO | CruisingPlugin | NONE | Planned maneuver from [24.79, 29.06) m/s over [19190.05, 19276.56) m 
 2018-06-19 10:53:37:753 | INFO | CruisingPlugin | NONE | Planned STEADY-SPEED maneuver at 29.06 m/s over [19276.56, 30569.19) m 
 2018-06-19 10:53:37:754 | INFO | Arbitrator | NONE | New trajectory planned in 9 ms. Planning finished at 0.007996015576228123% 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | Trajectory executor commanded to abort trajectory 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | TrajectoryExecutor received new trajectory! 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | Maneuver #1 from [19190.049137347014, 30569.189513248486) of type LATERAL 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | Maneuver #2 from [19190.049137347014, 19276.560727675307) of type LONGITUDINAL 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | Speeds from 24.785999298095703 to 29.0576 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | Maneuver #3 from [19276.560727675307, 30569.189513248486) of type LONGITUDINAL 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | Speeds from 29.0576 to 29.0576 
 2018-06-19 10:53:37:754 | INFO | TrajectoryExecutor | NONE | TrajectoryExecutor running trajectory! 
 2018-06-19 10:53:37:754 | INFO | ArbitratorStateMachine | NONE | Arbitrator processing event: FINISHED_TRAJECTORY_PLANNING 
 2018-06-19 10:53:37:755 | INFO | ArbitratorStateMachine | NONE | Arbitrator transitioned to state AWAITING_REPLAN
```

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 932

TonysCousin, 2020-04-17:
Applies to Carma2 architecture, so is now OBE.  Speed harm will now be implemented as speed advisories via geofence, which get applied in possibly long stretches to the world model.  This will no longer be handled by a plugin or arbitration.

closed 2020-04-17 by TonysCousin