usdot-fhwa-stol/carma-platform #123: Mobility messages need way of conveying non-path point information

msmcconnell, 2019-01-10:
In the TO26 cooperative merge application an RSU broadcasts a mobility request with the meter point. This request message is incorrectly processed as a vehicle path by mobility router causing numerous replans on all vehicles. PR #930  adds a band-aid fix for this by preventing replans for single point paths, but this demonstrates a need for mobility requests to convey messages without an included path. Perhaps a flag could be added to disable the concept of a path in a request message. Additionally, other messages types should probably support some kind of single point location information.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 931

msmcconnell, 2020-03-11:
This issue is still applicable to CARMA3. It boils down to the following.
Mobility Messages need to include an option to convey a request that is not a motion profile.