usdot-fhwa-stol/carma-platform #125: Platooning Gap is Abnormal

qswawrq, 2019-01-10:
During I-95 testing, we observed abnormal gaps when we are testing 0.84 time headway at 65mph.
The gap between vehicles from front bumper to rear bumper should be around 20 meters. But from radar data, it seems that the gap drops to 5~10 meters at some time for the third and forth vehicle.
[image]

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 926

TonysCousin, 2019-03-15:
I suggest we add logic to account for the time lag between message generation and message reading, rather than assuming instantaneous transmission.