usdot-fhwa-stol/carma-platform #127: Plugin params do not have unique prefix

msmcconnell, 2019-01-10:
The ros params used by plugins are all assigned to the private guidance_main namespace. However, unless the plugin specifies a prefix like example_plugin, then the parameters could clash and be overwritten. 

For example if a plugin has a timeout it will look like
/saxton_cav/guidance/guidance_main/timeout

but instead it we should make it
/saxton_cav/guidance/guidance_main/example_plugin/timeout

TonysCousin, 2019-03-11:
Formerly issue #918.

TonysCousin, 2019-03-18:
Per agreement with @msmcconnell this issue is now OBE, due to new architecture making plugins their own ROS nodes.  Each plugin can provide its own config file, if needed, and load it into the common config directory.

closed 2019-03-18 by TonysCousin