usdot-fhwa-stol/carma-platform #130: Logging IO Exception

qswawrq, 2019-01-10:
```
 2018-06-12 03:14:18:654 | SEVERE | Channel exception: java.io.IOException: Connection reset by peer 
 2018-06-12 03:14:18:659 | WARNING | GuidanceStateHandler | GUIDANCE_STATE | GuidanceStateHandlerreceived SHUTDOWN 
 2018-06-12 03:14:18:660 | INFO | GuidanceStateHandler | SHUTDOWN | Guidance state handler waiting5000ms for guidance thread shutdown... 
 2018-06-12 03:14:18:660 | INFO | GuidanceStateHandler | NONE | Publishing system_alert shutdown to ensure whole system shutdown 
 2018-06-12 03:14:18:663 | INFO | ManeuverInputs | NONE | Guidance.Maneuvers.ManeuverInputs shutting down normally. 
 2018-06-12 03:14:18:665 | INFO | GuidanceCommands | NONE | Guidance.Commands shutting down normally. 
 2018-06-12 03:14:18:671 | INFO | Tracking | NONE | Guidance.Tracking shutting down normally. 
 2018-06-12 03:14:18:673 | INFO | TrajectoryExecutor | NONE | Guidance.TrajectoryExecutor shutting down normally. 
 2018-06-12 03:14:18:688 | INFO | MobilityRouter | NONE | MobilityRouter shutting down normally. 
 2018-06-12 03:14:18:689 | INFO | LightBarManager | NONE | Light Bar Manager shutting down normally. 
 2018-06-12 03:14:18:689 | INFO | VehicleAwareness | NONE | VehicleAwareness shutting down normally. 
 2018-06-12 03:14:18:692 | INFO | I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused) 
 2018-06-12 03:14:18:692 | INFO | Retrying request 
 2018-06-12 03:14:18:692 | INFO | Arbitrator | NONE | Guidance.Arbitrator shutting down normally. 
 2018-06-12 03:14:18:692 | INFO | I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused) 
 2018-06-12 03:14:18:692 | INFO | Retrying request 
 2018-06-12 03:14:18:693 | INFO | I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused) 
 2018-06-12 03:14:18:693 | INFO | Retrying request 
 2018-06-12 03:14:18:694 | SEVERE | PluginManager | NONE | Guidance.PluginManager is about to SHUTDOWN! 
 2018-06-12 03:14:18:695 | WARNING | GuidanceStateHandler | GUIDANCE_STATE | GuidanceStateHandlerreceived SHUTDOWN 
 2018-06-12 03:14:18:699 | INFO | I/O exception (java.net.ConnectException) caught when processing request: Connection refused (Connection refused) 
```
This might be a new issue related to the new logging change. Need further investigate if it happens a lot.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 912

TonysCousin, 2020-04-17:
Applies to Carma2 architecture, so is now OBE.

closed 2020-04-17 by TonysCousin