usdot-fhwa-stol/carma-platform #131: Speed harm commands sometimes ignored on vehicle

TonysCousin, 2019-01-10:
During overnight testing on I-95 IHP the morning of 6/12, lead vehicle Green demonstrated at least one instance where it seemed to be ignoring incoming speed harm commands; it was receiving speed harm commands at ~22 m/s but GuidanceCommand was issuing commands to the controller at 29 m/s.  Quick scan of the logs seemed to indicate that the trajectory planning was ignoring speed harm, as no complex maneuver appeared in the trajectory. Not clear why.  One example is in green log 20180612-014837 at time 01:55:44.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 906

TonysCousin, 2020-04-17:
Applies to Carma2 architecture, so is now OBE.

closed 2020-04-17 by TonysCousin