usdot-fhwa-stol/carma-platform #132: Vehicle Immediately Stopping at the End of Route

maefromm, 2019-01-10:
On 5/22/2018 ATEF Testing, Phil (Volpe) said him and the drivers are concerned about the issue when the vehicles just suddenly stop at the end of route, especially when there's a platoon behind them. This doesn't happen all the time but when it does, it increases the likelihood of an accident. 

Attached are the 3 car platoon logs from today where Gray exhibited INACTIVITY and after re-engaging at the end of the route, it immediately stopped.

As a workaround, Tim said to split into two lanes towards the end of the route.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 880

TonysCousin, 2020-04-17:
Although this problem was with the Carma2 architecture, it may appear in a different manifestation in Carma3, so leaving it open for now as a reminder of things to watch for.

maefromm, 2021-12-03:
I have updated the title to reflect the issue that I think needs to be resolved which is the abrupt stopping behavior in vehicles. As of Battista testing, the trucks are also exhibiting the hard brake at the end of the route.