usdot-fhwa-stol/carma-platform #133: Multi-Vehicle Platooning requires vehicles to engage in order

qswawrq, 2019-01-10:
Current version of platooning plugin requires vehicle to engage one by one from the very first vehicle to the very last vehicle. The first two vehicle can engage in any order, but the third vehicle should engage when the first two vehicle have already formed a platoon (yellow light is on). The forth vehicle should wait to engage until the first three vehicle have already been in the same platoon.
The reason is that when a vehicle is searching for a platoon, it only cares if the rear position of any platoon is in front of the host vehicle, which means that it will not care about whether there is an object between the host vehicle and the front platoon.
In three vehicle case, if they started in random order, there will be a case where the third vehicle is joining the first vehicle. And the second vehicle is never able to join them because it is between the first and the third.
#827 is a fix for that, but the sensor fusion anomaly prevent it from working properly. (See #842)

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 873

TonysCousin, 2020-04-17:
While this applied to Carma2, and the architecture has significantly changed, we will be re-introducing the same APF platooning logic, so this issue may still be with us.

qswawrq, 2020-04-17:
@TonysCousin I confirmed it is still a valid issue and will exist with our CARMA3 implementation.