usdot-fhwa-stol/carma-platform #135: Platooning Plugin depends on the same route file

qswawrq, 2019-01-10:
Current implementation of platooning is letting each vehicle report their own downtrack distance. It would cause a failure in some cases:
1. If two vehicles are using two different route files, they will have different starting points for the downtrack distance.
2. If two vehicles are using the same route file, but one vehicle join the route before it is actually on any route segment, that vehicle will have a different starting point with the vehicle who join the route when it is already on the route.
In both cases, for the same location, they will report different dtd, which means they should not platoon together.
The current workaround is to avoid those two cases. But it should be fixed when the sensor fusion is functionally ready. See #842

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 866

TonysCousin, 2020-04-17:
Only applies to Carma2 architecture, so this is now OBE.

closed 2020-04-17 by TonysCousin