usdot-fhwa-stol/carma-platform #136: Ramp Metering - Stopping Issue

maefromm, 2019-01-10:
Takes minutes for "merge" vehicle to stop by coasting to 3MPH, and jerks until then. 

Give TORC an email/call on controlling a vehicle to a stop. This is potentially a scope creep but try to address now if possible. Assign to Ed to contact Tammy R. to resolve this somehow and have Mike to followup.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 859

msmcconnell, 2019-03-11:
The firmware on all the Cadillac SRX's was updated to address this issue. 
The behavior is now as follows.
The vehicle will bring itself to a stop and hold when commanded 0 m/s and some non-zero max acceleration. 
To release the vehicle the user must command a non-zero speed and non-zero max acceleration. Additionally, after the commands are sent the user must briefly tap the accelerator to allow the vehicle to continue moving.

closed 2019-03-11 by msmcconnell