usdot-fhwa-stol/carma-platform #137: Platoon Vehicles - Inactive state

maefromm, 2019-01-10:
Notes from our Task Meeting 5/14/2018:

Inactive State - 220ms controller timeouts which may be caused by volume of messages. Operations message with command speeds. With BSMs and Path messages off, there was only 3 out of 4 that was inactive. Speed cmds are always sent at 100ms, regardless if the message is received or not. It takes 0.001s Message to Mobility. There is 0.3s delay from MobilityRouter to Platooning. It's a ROS delay. 

TODO: Need to ask Tim to create a double lap route file with 4-5 vehicles next week.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 857

TonysCousin, 2020-04-17:
Only applies to Carma2 architecture, so this is now OBE.

closed 2020-04-17 by TonysCousin