usdot-fhwa-stol/carma-platform #138: Sensor fusion reports connected vehicle in a wrong lane

qswawrq, 2019-01-10:
When we are doing two-vehicle platooning testing, the first vehicle is in the same lane of the second one broadcasting BSM. The seconds vehicle see one object in front with empty BSM id and see another vehicle with BSM in a different lane (lane 4). Here are logs file for the second vehicle:
[F_silver_test1.txt](https://github.com/fhwa-saxton/CarmaPlatform/files/1988608/F_silver_test1.txt)
The log looks like this:
```2018-05-08 07:30:43:873 | DEBUG | PlatooningPlugin | NONE | Found objects from roadway:
2018-05-08 07:30:43:873 | DEBUG | PlatooningPlugin | NONE | Found obstacle in lane 0 and its dtd is 226.15253250848778
2018-05-08 07:30:43:874 | DEBUG | PlatooningPlugin | NONE | The found obstacle type is MessageImpl<cav_msgs/ConnectedVehicleType>
2018-05-08 07:30:43:874 | DEBUG | PlatooningPlugin | NONE | The BSM id is of size 0
2018-05-08 07:30:43:874 | DEBUG | PlatooningPlugin | NONE | Found obstacle in lane 4 and its dtd is 187.2083069788181
2018-05-08 07:30:43:874 | DEBUG | PlatooningPlugin | NONE | The found obstacle type is MessageImpl<cav_msgs/ConnectedVehicleType>
2018-05-08 07:30:43:874 | DEBUG | PlatooningPlugin | NONE | The BSM id is of size 4
2018-05-08 07:30:43:874 | DEBUG | PlatooningPlugin | NONE | The content at 0 is 73
2018-05-08 07:30:43:875 | DEBUG | PlatooningPlugin | NONE | The content at 1 is 54
2018-05-08 07:30:43:875 | DEBUG | PlatooningPlugin | NONE | The content at 2 is -59
2018-05-08 07:30:43:875 | DEBUG | PlatooningPlugin | NONE | The content at 3 is -70
```
I confirmed that the reported BSM ID in lane 4 is what the first vehicle is boardcasting.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 842

TonysCousin, 2020-04-17:
Only applies to Carma2 architecture, so this is now OBE.

closed 2020-04-17 by TonysCousin