usdot-fhwa-stol/carma-platform #140: Speedharm Plugin should not throw Fatal when SpeedHarm Server is unresponsive

maefromm, 2019-01-10:
While testing on local VM, got this Fatal Error when :
1. carma UI running and engaged while speed harm server running. 
2. speed harm is receiving vehiclecommands and carma UI receiving SH commands.
3. stopped speed harm server (which is similar to server not being available during a run) 
4. On the UI, driver disengages and hits continue. Then carma throws a fatal error.  

Expected behavior should be to make the plugin unavailable and continue with the other plugins. 

2018-05-03 16:59:58:768 | INFO | PluginLifecycleHandler | PLUGIN | Suspending Cruising Plugin:1.0.2
2018-05-03 16:59:58:774 | DEBUG | YieldPlugin | NONE | Availability Listeners 1
2018-05-03 16:59:58:777 | INFO | PluginLifecycleHandler | PLUGIN | Suspending Yield Plugin:1.0.1
2018-05-03 16:59:58:777 | INFO | LightBarManager | NONE | Set light bar for Light Bar State Machine for LightBarIndicator GREEN with statusOFF
2018-05-03 16:59:58:778 | INFO | LightBarStateMachine | NONE | Event: GUIDANCE_DISENGAGED Prev State: ENGAGED New State: DISENGAGED
2018-05-03 16:59:58:778 | DEBUG | LightBarManager | NONE | Light Bar Manager is polling job queue.
2018-05-03 16:59:58:778 | FATAL | GuidanceMain | NONE | Guidance thread Speed Harmonization PluginDoSuspendPluginWorkerThread raised uncaught exception! Handling!!!
2018-05-03 16:59:58:778 | ERROR | GuidanceExceptionHandler | NONE | Global guidance exception handler caught an uncaught exception from thread: Speed Harmonization PluginDoSuspendPluginWorkerThread
org.springframework.web.client.ResourceAccessException: I/O error on DELETE request for "http://localhost:8081/rest/vehicles/15": Connection refused (Connection refused); nested exception is java.net.ConnectException: Connection refused (Connection refused)
	at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:728)
	at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:671)
	at org.springframework.web.client.RestTemplate.delete(RestTemplate.java:540)
	at gov.dot.fhwa.saxton.carma.plugins.speedharm.SessionManager.endVehicleSession(SessionManager.java:81)
	at gov.dot.fhwa.saxton.carma.plugins.speedharm.SpeedHarmonizationPlugin.onSuspend(SpeedHarmonizationPlugin.java:176)
	at gov.dot.fhwa.saxton.carma.guidance.plugins.SuspendPluginTask.run(SuspendPluginTask.java:29)
	at gov.dot.fhwa.saxton.carma.guidance.plugins.PluginLifecycleHandler$PluginWorker.run(PluginLifecycleHandler.java:299)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
	at sun.net.www.http.HttpClient.New(HttpClient.java:339)
	at sun.net.www.http.HttpClient.New(HttpClient.java:357)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1220)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:984)
	at org.springframework.http.client.SimpleBufferingClientHttpRequest.executeInternal(SimpleBufferingClientHttpRequest.java:78)
	at org.springframework.http.client.AbstractBufferingClientHttpRequest.executeInternal(AbstractBufferingClientHttpRequest.java:48)
	at org.springframework.http.client.AbstractClientHttpRequest.execute(AbstractClientHttpRequest.java:53)
	at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:714)
	... 7 more
2018-05-03 16:59:58:779 | DEBUG | GuidanceStateMachine | GUIDANCE_STATE | Guidance state machine reveiced PANIC at state: DRIVERS_READY
2018-05-03 16:59:58:779 | DEBUG | GuidanceStateMachine | GUIDANCE_STATE | Guidance transited to state SHUTDOWN
2018-05-03 16:59:58:779 | DEBUG | GuidanceStateMachine | GUIDANCE_STATE | Guidance is taking action PANIC_SHUTDOWN
2018-05-03 16:59:58:782 | DEBUG | LaneChangePlugin | NONE | Availability Listeners 1
2018-05-03 16:59:58:784 | INFO | PluginLifecycleHandler | PLUGIN | Suspending Lane Change Plugin:1.0.2
2018-05-03 16:59:58:785 | INFO | YieldPlugin | NONE | Yield plugin suspended
2018-05-03 16:59:58:787 | DEBUG | RouteFollowingPlugin | NONE | Availability Listeners 1

******

WARNING:  | CommandReceiver | NONE | Unable to wait for server speed command, received exception.
org.springframework.web.client.ResourceAccessException: I/O error on GET request for "http://localhost:8081/rest/commands/8": Connection refused (Connection refused); nested exception is java.net.ConnectException: Connection refused (Connection refused)
	at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:728)
	at org.springframework.web.client.RestTemplate.execute(RestTemplate.java:671)
	at org.springframework.web.client.RestTemplate.getForObject(RestTemplate.java:319)
	at gov.dot.fhwa.saxton.carma.plugins.speedharm.CommandReceiver.run(CommandReceiver.java:53)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
[... 46 lines cut ...]
	at gov.dot.fhwa.saxton.carma.guidance.plugins.PluginLifecycleHandler$PluginWorker.run(PluginLifecycleHandler.java:299)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
	at java.net.PlainSocketImpl.socketConnect(Native Method)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at java.net.Socket.connect(Socket.java:538)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:180)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
	at sun.net.www.http.HttpClient.New(HttpClient.java:339)
	at sun.net.www.http.HttpClient.New(HttpClient.java:357)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1220)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:984)
	at org.springframework.http.client.SimpleBufferingClientHttpRequest.executeInternal(SimpleBufferingClientHttpRequest.java:78)
	at org.springframework.http.client.AbstractBufferingClientHttpRequest.executeInternal(AbstractBufferingClientHttpRequest.java:48)
	at org.springframework.http.client.AbstractClientHttpRequest.execute(AbstractClientHttpRequest.java:53)
	at org.springframework.web.client.RestTemplate.doExecute(RestTemplate.java:714)
	... 7 more

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 826

maefromm, 2020-03-09:
per @TonysCousin, we will close this out on Github as part of CARMA2 and will have separate documentation of old issues on SharePoint.

closed 2020-03-09 by maefromm