usdot-fhwa-stol/carma-platform #148: Gap display

TonysCousin, 2019-01-10:
Tim Tiernan has requested that the UI always display the sensed gap if a forward vehicle is in view in host's lane (show distance in meters).  Ideally, this would show just the track 1 data from the Cadillac radar, but if we can't easily get that isolated, then whatever data we have coming out of sensor fusion.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 769

maefromm, 2020-09-28:
@dan-du-car I believe this is already implemented on the updated basic travel ui for platooning info, correct?

dan-du-car, 2020-09-28:
@maefromm   This line https://github.com/usdot-fhwa-stol/carma-web-ui/blob/6ed063bfb9b98c44eba6a493bd35a0e287d654b0/website/scripts/ros/ros_platooning_info.js#L66 display the platoonInfo. There is a desired gap. Actual gap is not published yet. I am not sure which gap is the correct data.