usdot-fhwa-stol/carma-platform #150: Longitudinal maneuvers should allow controller to define speed profile

TonysCousin, 2019-01-10:
Current implementations of the speedup and slowdown maneuvers compute a linear interpolation between start speed and end speed then feed that interpolated command to the controller on every time step.  The controller would perform better if we gave it the final target and let its PID naturally produce the critically damped response.  But because of the controller's need for an input every 200 ms we need to consider whether a frequent re-issuance of the target command is any better than providing the linear command change at each time step. @watters-torc , your input would be valuable here.
This is not a critical, short-term need, but something to consider longer term.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 647

TonysCousin, 2020-04-17:
Applies only to the Carma2 architecture, so is now OBE.

closed 2020-04-17 by TonysCousin