usdot-fhwa-stol/carma-platform #154: No indication of E-Stop being pressed. (Aberdeen SpeedHarm Test)

msmcconnell, 2019-01-10:
A GoPro camera fell from its ceiling mounted position and hit the E-Stop button. This resulted in a system shutdown when the controller reported a FAULT. However discovering the source of this issue was time consuming as no one has observed the button being pressed. Some indication of E-Stop compression needs to be available in the vehicle such as an LED indicator.

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 416

msmcconnell, 2020-03-11:
This is still an issue in some of the CARMA3 vehicles. Some of the E-stops have an led indicator but others do not.

TonysCousin, 2020-04-17:
It would be nice to have an indicator on the Carma UI to show the state of the E-stop.

maefromm, 2020-05-18:
@msmcconnell what existing topic that can be used to determine the e-stop has been pressed. Does the UI need to listen to ssc controller?