usdot-fhwa-stol/carma-platform #161: Wifi capability on vehicle computers

TonysCousin, 2019-01-10:
If vehicles are to use I2 connection at Aberdeen they will need to connect to the wifi signal there. We need to get wifi adapters/dongles for each computer.  Whether we use this method or direct ethernet cable (not sure if that will be available at Aberdeen), we will need careful and foolproof procedures in place to be sure that the vehicle computer is never connected to I2 and another network (e.g. Verizon aircard) simultaneously.  Having a dongle that uses the same USB slot as an aircard would be a good way to prevent accidentally using them both (block other slots).

darrelld05, 2019-01-10:
This issue was transferred from the original CARMAPlatform: Old issue # 212

msmcconnell, 2022-01-14:
Closing as this is no longer an objective in the foreseeable future.

closed 2022-01-14 by msmcconnell