usdot-fhwa-stol/carma-platform #195: InterfaceWorker: missing one or more essential drivers

maefromm, 2019-02-27:
### Descriptive summary
Unable to find essential drivers. Potential issue coming from recent changes on Interface Manager or launch file may need to be updated. 

### Expected behavior
CARMA platform should not run into this error. 

### Actual behavior
Error during manual launch on command line using remote-install.bash. 

### Steps to reproduce the behavior

1. remote-install.bash deployment.
2. manually launch using command line.. 

### Related work
NA

### Logs
[rosout.log](https://github.com/usdot-fhwa-stol/CARMAPlatform/files/2911736/rosout.log)

qswawrq, 2019-02-27:
Previously, our drivers publish the driver status topic on "driver_discovery" and this string is mapped to "saxton_cav/drivers/driver_discovery". Due to some recent changes on drivers code, "driver_discovery" string is mapped to "/driver_discovery" topic name, but InterfaceManager is still subscribing to the original topic "saxton_cav/drivers/driver_discovery". The fix should be correcting the topic mapping for all our drivers.

maefromm, 2019-03-26:
@Supraluminal when running CARMA without docker, below is the output of the rostopic info. As you can see there are 2 sets of driver_discovery. I assume the proper fix is to remap the drivers to use "/saxton_cav/drivers/driver_discovery " instead of just "/drivers_discovery". If so 2 potential fixes, UNLESS this is kinda null/void already since by the time we fix this, your docker development deployment script would already be working order?  Anyways here they are: 
1. Update all the driver's launch file (e.g. pinpoint/launch/pinpoint.launch) to force or ensure it's mapped/remapped to "/saxton_cav/drivers/driver_discovery" 
2. Update "saxton_cav_docker.launch" or have it call a different version of "carmajava/launch/saxton_cav_src.launch" to use "<remap from="driver_discovery" to="/driver_discovery"/>" when not using docker. Right now the manual fix is to update the saxton_cav_src.launch. 

--------- ROSTOPIC INFO ---------
maefromm@carma-pc:~$ rostopic info /driver_discovery
Type: cav_msgs/DriverStatus
Publishers:
 * /dsrc (http://192.168.88.10:44681/)
 * /srx_objects (http://192.168.88.10:42871/)
 * /srx_can (http://192.168.88.10:44151/)
 * /pin_point_application (http://192.168.88.10:39703/)
 * /srx_controller (http://192.168.88.10:45437/)
Subscribers:
 * /saxton_cav/rosbag_node (http://192.168.88.10:38109/)
 
 
maefromm@carma-pc:~$ rostopic info /saxton_cav/drivers/driver_discovery
Type: cav_msgs/DriverStatus
Publishers:
 * /saxton_cav/drivers/lateral_controller (http://192.168.88.10:36645/)
Subscribers:
 * /saxton_cav/interface_manager/interface_mgr (http://192.168.88.10:44773/)
 * /saxton_cav/rosbag_node (http://192.168.88.10:38109/)

TonysCousin, 2019-07-29:
Appears to be limited to CARMA2.

TonysCousin, 2019-10-24:
No further activity or concern, and CARMA2 limitations, so closing as OBE.

closed 2019-10-24 by TonysCousin