usdot-fhwa-stol/carma-platform #198: PidControllerTest fails on CI environment

qswawrq, 2019-03-05:
### Descriptive summary
PidControllerTest unit test failed on CI environment once or twice (very rare).
### Expected behavior
All unit test should pass deterministically.
### Actual behavior
PidControllerTest might fail in some rare cases.
### Steps to reproduce the behavior
Hard to reproduce. If you run CI build for many times, it might show up.
### Related work
#184 
Both unit tests are using system time. It causes the issue on CI environment.

TonysCousin, 2019-10-24:
Associated user story CAR-444 has been closed out since this only applies to Cadillacs, which are at end of life.  Need to consider whether this anomaly report should also be closed.

TonysCousin, 2020-04-17:
No plans to further support Cadillacs, so closing this as OBE.

closed 2020-04-17 by TonysCousin