usdot-fhwa-stol/carma-platform #200: Index out of bounds exception in platooning docker demo

msmcconnell, 2019-03-11:
### Descriptive summary
While testing the docker platooning demo config an index out of bounds exception was generated resulting in system alert fatal.

```

java.lang.IndexOutOfBoundsException: Index: -1, Size: 0
at java.util.LinkedList.checkElementIndex(LinkedList.java:555)
at java.util.LinkedList.get(LinkedList.java:476)
at gov.dot.fhwa.saxton.carma.guidance.util.trajectoryconverter.TrajectoryConverter.convertToPath(TrajectoryConverter.java:220)
at gov.dot.fhwa.saxton.carma.guidance.util.trajectoryconverter.TrajectoryConverter.convertToPath(TrajectoryConverter.java:158)
at gov.dot.fhwa.saxton.carma.guidance.VehicleAwareness.getPathPrediction(VehicleAwareness.java:171)
at gov.dot.fhwa.saxton.carma.guidance.VehicleAwareness.getPathPrediction(VehicleAwareness.java:154)
at gov.dot.fhwa.saxton.carma.guidance.VehicleAwareness.notifyNewTrajectoryPlanned(VehicleAwareness.java:201)
at gov.dot.fhwa.saxton.carma.guidance.arbitrator.Arbitrator.complexTrajectoryReplan(Arbitrator.java:543)
at gov.dot.fhwa.saxton.carma.guidance.GuidanceComponent.runJobQueue(GuidanceComponent.java:135)
at gov.dot.fhwa.saxton.carma.guidance.GuidanceComponent$1.loop(GuidanceComponent.java:154)
at org.ros.concurrent.CancellableLoop.run(CancellableLoop.java:56)
at java.lang.Thread.run(Thread.java:748)

```

### Expected behavior
No exceptions should appear during normal operation
### Actual behavior
Index out of bounds exception thrown
### Steps to reproduce the behavior

Run the platooning docker demo with 3 vehicles on colonial farm rode using version 2.8.3

### Related work

This corresponds to JIRA issue CAR-491

msmcconnell, 2019-03-11:
Investigation of the logs shows that this issue is caused by a post complex maneuver steadying trajectory being requested by the arbitrator.
```
 2019-02-21 15:44:37:798 | INFO | ArbitratorStateMachine | NONE | Arbitrator transitioned to state REPLAN_AFTER_COMPLEX_TRAJECTORY 
 2019-02-21 15:44:37:799 | INFO | Arbitrator | NONE | Arbitrator replanning trajectory after complex maneuver execution 
 2019-02-21 15:44:37:799 | INFO | CruisingPlugin | NONE | Found 1 empty gaps in trajectory [401.9034641049786, 434.09346273168757] 
 2019-02-21 15:44:37:799 | INFO | CruisingPlugin | NONE | Gap: [startLocation = 401.9034641049786; endLocation = 434.09346273168757; startSpeed = 6.7056] 
 2019-02-21 15:44:37:799 | INFO | CruisingPlugin | NONE | Found 2 speed limits in Gap: [startLocation = 401.9034641049786; endLocation = 434.09346273168757; startSpeed = 6.7056] 
 2019-02-21 15:44:37:799 | INFO | CruisingPlugin | NONE | SpeedLimit: [location = 401.9034641049786, limit = 6.7056] 
 2019-02-21 15:44:37:799 | INFO | CruisingPlugin | NONE | SpeedLimit: [location = 434.09346273168757, limit = 15.6464] 
```
When cruising tries to plan this trajectory it returns 2 speed limits which do not match those provided by the route service on startup
```
 2019-02-21 15:43:08:832 | INFO | LocalSpeedLimitConstraint | NONE | Loaded: SpeedLimit={location=355.48, limit=15.65} 
 2019-02-21 15:43:08:833 | INFO | LocalSpeedLimitConstraint | NONE | Loaded: SpeedLimit={location=379.99, limit=15.65} 
 2019-02-21 15:43:08:834 | INFO | LocalSpeedLimitConstraint | NONE | Loaded: SpeedLimit={location=401.90, limit=15.65} 
 2019-02-21 15:43:08:835 | INFO | LocalSpeedLimitConstraint | NONE | Loaded: SpeedLimit={location=429.58, limit=15.65} 
 2019-02-21 15:43:08:837 | INFO | LocalSpeedLimitConstraint | NONE | Loaded: SpeedLimit={location=451.18, limit=15.65} 
```
This has two root causes there is a bug in the GuidanceRouteService.getSpeedLimitAtLocation function where the last limit on the route can be returned prematurely
```
if (location > limits.last().getLimit()) {	
      return limits.last();	
    }
```
This should say location > limits.last().getLocation()The second root cause is that platooning is inadvertently modifying the speed limit list after a call to routeService.getSpeedLimitsInRange()
```
            // Plan a speed up maneuver to speed limit and a steady speed maneuver at the speed limit
            // TODO the following logic is from cruising plugin, I feel uncomfortable to re-use it here
            // TODO the following logic should be moved to route service
            SortedSet<SpeedLimit> trajLimits = rs.getSpeedLimitsInRange(traj.getStartLocation(), traj.getEndLocation());
            trajLimits.add(rs.getSpeedLimitAtLocation(traj.getEndLocation()));
            // Merge segments with same speed limits
            List<SpeedLimit> mergedLimits = new LinkedList<SpeedLimit>();
            SpeedLimit limit_buffer = null;
            for (SpeedLimit limit : trajLimits) {
               // Merge segments with same speed
                if (limit_buffer == null) {
                    limit_buffer = limit; // !!!! OFFENDING LINE !!!!
                } else {
                    if (Math.abs(limit_buffer.getLimit() - limit.getLimit()) < SPEED_EPSILON) {
                        limit_buffer.setLocation(limit.getLocation());
                    } else {
                        mergedLimits.add(limit_buffer);
                        limit_buffer = limit; // !!!! OFFENDING LINE !!!!
                    }
                }
            }
            limit_buffer.setLocation(traj.getEndLocation());
            mergedLimits.add(limit_buffer);
```
Since these are small changes, I have put up a PR https://github.com/usdot-fhwa-stol/CARMAPlatform/pull/201 to address them.

qswawrq, 2019-03-12:
This bug should be addressed.

closed 2019-03-12 by qswawrq