usdot-fhwa-stol/carma-platform #202: Security risk on incoming mobility messages

TonysCousin, 2019-03-22:
### Descriptive summary
Feb 2019 SonarCloud scan of the code base identified a security hotspot where a regex is being used to scan mobility message strategy strings, and allowing anything in that is between commas (splitting it into comma-separated key/value pairs).  Investigation of this logic led to several places in RSUMetering and CooperativeMerge plugins where these values are used unchecked.  Some of the values have the potential to drastically affect how the vehicle behaves.

### Expected behavior
The plugins should do a sanity check on the incoming strategy values to improve confidence that they are legitimate.

### Actual behavior
Received strategy values are used as-is with no verification.

### Steps to reproduce the behavior
Software execution is not necessary.  Review Sonarcloud scan from Feb 2019 to see the item (https://sonarcloud.io/project/issues?id=usdot-fhwa-stol_CARMAPlatform&open=AWi6MsbKa3c5WIT7ymyM&resolved=false&severities=CRITICAL&types=VULNERABILITY)

### Related work

None.

darrelld05, 2019-04-17:
removed the label

msmcconnell, 2019-05-24:
@TonysCousin Has this been resolved?

msmcconnell, 2020-03-11:
My understanding from PR #212 is that this issue is resolved. Will close.

closed 2020-03-11 by msmcconnell