usdot-fhwa-stol/carma-platform #205: Address Sonarcloud Security Vulnerabilities

qswawrq, 2019-04-04:
### Descriptive summary
Based on Sonarcloud report, our CARMA platform have 54 security vulnerabilities:
https://sonarcloud.io/project/issues?id=usdot-fhwa-stol_CARMAPlatform&open=AWi6MsbKa3c5WIT7ymyM&resolved=false&types=VULNERABILITY
### Expected behavior
CARMA platform should pass Sonarcloud code vulnerability check.
### Actual behavior
Sonarcloud found 54 security issues.
### Steps to reproduce the behavior
No need to reproduce. See Sonarcloud report for detail.
### Related work
N/A

msmcconnell, 2019-05-24:
@qswawrq @TonysCousin Has this been completed?

TonysCousin, 2019-05-24:
Yes, I believe it has.  There was one defect in all 54, and it was handled under another issue.  Want to give @qswawrq a chance to comment before closing, though.

msmcconnell, 2020-03-11:
Enough time has passed this issue can be closed. Vulnerabilities found/created after the May 24 2019 data above will be handled in their own issues as necessary.

closed 2020-03-11 by msmcconnell