usdot-fhwa-stol/carma-platform #213: New version of CARMAMsg is not compatible with the current CARMAPlatform

qswawrq, 2019-04-19:
### Descriptive summary
CircleCI fails on the latest develop branch.
```
/opt/carma/src/CARMAPlatform/carmajava/route/src/main/java/gov/dot/fhwa/saxton/carma/route/RouteWaypoint.java:92: error: cannot find symbol
      newManeuver.setLength(maneuver.getLength());
                                    ^
  symbol:   method getLength()
  location: variable maneuver of type Maneuver
/opt/carma/src/CARMAPlatform/carmajava/route/src/main/java/gov/dot/fhwa/saxton/carma/route/RouteWaypoint.java:93: error: cannot find symbol
      newManeuver.setPerformers(maneuver.getPerformers());
                                        ^
  symbol:   method getPerformers()
  location: variable maneuver of type Maneuver
/opt/carma/src/CARMAPlatform/carmajava/route/src/main/java/gov/dot/fhwa/saxton/carma/route/RouteWaypoint.java:94: error: cannot find symbol
      newManeuver.setStartRoadwayLaneId(maneuver.getStartRoadwayLaneId());
                                                ^
  symbol:   method getStartRoadwayLaneId()
  location: variable maneuver of type Maneuver
/opt/carma/src/CARMAPlatform/carmajava/route/src/main/java/gov/dot/fhwa/saxton/carma/route/RouteWaypoint.java:95: error: cannot find symbol
      newManeuver.setStartRoadwayLink(maneuver.getStartRoadwayLink());
                                              ^
  symbol:   method getStartRoadwayLink()
  location: variable maneuver of type Maneuver
/opt/carma/src/CARMAPlatform/carmajava/route/src/main/java/gov/dot/fhwa/saxton/carma/route/RouteWaypoint.java:96: error: cannot find symbol
      newManeuver.setStartRoadwayOriginatorPosition(maneuver.getStartRoadwayOriginatorPosition());
                                                            ^
  symbol:   method getStartRoadwayOriginatorPosition()
  location: variable maneuver of type Maneuver
5 errors

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':route:compileJava'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.
CARMAPlatform/carmajava/CMakeFiles/gradle-carma.dir/build.make:57: recipe for target 'CARMAPlatform/carmajava/CMakeFiles/gradle-carma' failed
make[2]: *** [CARMAPlatform/carmajava/CMakeFiles/gradle-carma] Error 1
CMakeFiles/Makefile2:12254: recipe for target 'CARMAPlatform/carmajava/CMakeFiles/gradle-carma.dir/all' failed
make[1]: *** [CARMAPlatform/carmajava/CMakeFiles/gradle-carma.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make install -j36 -l36" failed
Exited with code 1
```
But the code can be build locally with an old version of CARMAMsg package.
### Expected behavior
CircleCI should pass always.
### Actual behavior
CircleCI fails.
### Steps to reproduce the behavior
Build CARMAPlatform with the latest CARMAMsg package.
### Related work
https://github.com/usdot-fhwa-stol/CARMAMsgs/pull/7 breaks the depency between them.

Supraluminal, 2019-04-19:
What's the commit hash for the older version of CARMAMsgs you used to successfully build locally? The `docker/checkout.sh` script can be modified to use that commit very easily which should fix the issue.

Supraluminal, 2019-04-22:
Actually, upon further investigation, I'm not even sure why the offending lines of code are even there. They're setting up a Maneuver message (which I don't think we had before) that is then never stored anywhere or used ever again. My proposed solution would be to delete those lines entirely. They just seem to be cruft left over from probably **early** CARMA2 development that haven't had a purpose in a long time.

PR #214 (2019-04-29): Remove obsolete references to deprecated maneuver messages
# PR Details
Fixes issues #213 by removing old references to a former concept of maneuvers that is now being surfaced due to changes in `cav_msgs/Maneuver.msg`.

## Description

By modifying `cav_msgs/Maneuver.msg` this old (apparently unused and unneeded) code was broken. The removed code referenced a deep copy of a set of required maneuvers which was never used and never even stored inside the object that was doing the copy. As such this code has been removed. It may be worth considering removing this field from `cav_msgs/RouteWaypoi [... cut]
closed 2019-04-29 by qswawrq