usdot-fhwa-stol/carma-platform #222: TrafficSignalPlugin stopping case not robust to signal loss; JIRA Issue: #CAR2-106

darrelld05, 2019-05-06:
### Descriptive summary
While testing a stop case in TSP I discovered that signal to the current intersection appeared to be lost while the car was stopped. This caused the car to start driving again. This issue needs to be investigated further.log 102439 is the log from the failed run. log 103415 was a test right after conducted on the same branch/commit but did not experience the issue.
This task is complete if a cause for the issue was identified and a fix implemented and regression tested.

maefromm, 2019-12-18:
@msmcconnell closing out the JIRA task due to "CARMA2 closeout. Most likely OBE or will revisit if need in CARMA3" Would you like to keep this open?

maefromm, 2019-12-18:
@msmcconnell. One idea is to probably put all the issues/anomalies with CARMA2 TSP into the initial CARMA3 Design Documentation for TSP user story, to consider all these issues we've previously run into. And then close this issue until it comes back up again in CARMA3. Do you agree?

closed 2020-03-11 by msmcconnell