usdot-fhwa-stol/carma-platform #252: Autoware message package dependencies missing

Supraluminal, 2019-06-06:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
```-- Could not find the required component 'autoware_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "autoware_msgs"
  with any of the following names:

    autoware_msgsConfig.cmake
    autoware_msgs-config.cmake

  Add the installation prefix of "autoware_msgs" to CMAKE_PREFIX_PATH or set
  "autoware_msgs_DIR" to a directory containing one of the above files.  If
  "autoware_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CARMAPlatform/pure_pursuit_wrapper/CMakeLists.txt:13 (find_package)
```

during Docker build. Suspect that Autoware message packages need to be baked into a new CARMABase release and the Docker build needs to be updated accordingly.

### Expected behavior
Docker build completes successfully.
### Actual behavior
Docker build errors out with the above error message.
### Steps to reproduce the behavior

1. Checkout CARMAPlatform repository
1. Run `./docker/build-image.sh`

msmcconnell, 2019-06-06:
@Supraluminal This is because you need to use the carma_build script now to build the platform. I haven't had a chance to send out an update about it yet.

qswawrq, 2019-06-18:
This is fixed by #258 .

closed 2019-06-18 by qswawrq