usdot-fhwa-stol/carma-platform #300: Exception at end of route in CARMA Pre-Release

msmcconnell, 2019-07-26:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Hit the brake just before the end of the TFHRC demo route in CARMA 3.0 pre-release. Shortly after received and exception (see below).
[image]

### Expected behavior
Car brings itself to a stop at end of route and braking out does not result in an exception
### Actual behavior
Exception shown when braking out near end of route sometimes.

msmcconnell, 2019-07-26:
Logs from the blue lexus
[exception_logs_.zip](https://github.com/usdot-fhwa-stol/CARMAPlatform/files/3436857/exception_logs_.zip)

The rosbag is still on the blue lexus at /opt/carma/logs/_2019-07-25-15-03-48.bag

maefromm, 2019-08-29:
@TonysCousin as discussed, this issue will include the issue with the hard braking at the end. As a workaround we can fix the speed profile on the waypoints for the TFHRC Circle.

TonysCousin, 2019-09-09:
Per group discussion, Guidance needs to recognize end of route and present that info to the user.  Don't allow it to just run out of route data, for which situation the shown exception is the desired behavior.  This should be taken care of once the remainder of Guidance is completed.

maefromm, 2019-09-28:
@TonysCousin  In regards to fixing the speed profile as a workaround to not have CARMA abruptly brake hard at the end of the route, it did not work. 
- No matter how Saina and I adjust the waypoints at the end of the route, CARMA does the following:
1) Abruptly brakes hard. 
2) Stops a lot sooner than Autoware does. 
Depending if the exception can include the fix for the 2 issues above. If not, then we should create a separate issue to address these.

maefromm, 2019-10-08:
Per @qswawrq the RouteEvent "ROUTE COMPLETED" is not implemented and therefore UI does not provide the popup to notify user that it has reached end of route. Guidance needs to re-implement this or something similar that will notify the UI that route has been completed.

maefromm, 2019-10-09:
As discussed on email, created a separate ticket for the slamming of the brakes #383

maefromm, 2019-12-04:
Decided during Sprint 21 Planning that we are postponing this until Guidance and Route Manager completely handles the end of route.

maefromm, 2020-09-28:
@TonysCousin @kjrush @msmcconnell : I think this is OBE since with Vanden Plas, this whole logic will be overriden with newest logic/code. We can re-open if the same error still comes up with the new code.

msmcconnell, 2021-01-21:
I am going to close this as OBE per Mae's Sep 28th 2020 comment. 
Additionally, Anish recently demonstrated functionality to bring the vehicle to a stop at the end of the route, so this would seem to no longer be an issue.

msmcconnell, 2021-01-21:
Nevermind. Reopening per Mae's Oct 2020 comment on Jira.

msmcconnell, 2021-03-22:
Testing of 3.5.0 at turner showed that when the vehicle manages to stop under control of the stop and wait plugin then this issue does not occur. There are separate issues open related to the reliability of stop and wait to bring the vehicle to a stop.

closed 2021-03-22 by msmcconnell