usdot-fhwa-stol/carma-platform #302: Autoware package missing in carma_build script

msmcconnell, 2019-07-26:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The waypoint_planner package is missing in the carma_build script but is used in the launch files. So CARMA will only run if a separate autoware installation containing that package is found.
### Expected behavior
Using only the CARMA repos and required ros dep packages a user can build and run carma
### Actual behavior
An additional autoware source installation is required
### Steps to reproduce the behavior
Clone carma repos and build using carma_build then try to launch

TonysCousin, 2019-07-26:
For the record here, reference to "carma repos" above includes the carma fork of autoware.

TonysCousin, 2019-07-26:
Closing since PR #303 has been approved & merged.

closed 2019-07-26 by TonysCousin