usdot-fhwa-stol/carma-platform #316: Update repo for conform with parameter standards

msmcconnell, 2019-08-02:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
This repo should be updated to conform to the parameters standards document found here https://usdot-carma.atlassian.net/wiki/spaces/CAR/pages/196182019/CARMA3+Project+Documentation?preview=/196182019/198574135/CARMA%20Platform%20Detailed%20Design%20-%20Parameter%20and%20Launch%20Standards%20for%20Different%20Vehicle%20Configurations.docx

TonysCousin, 2019-09-05:
@msmcconnell I have looked through the referenced standards doc above and do not see any obvious guidance there about how to structure a repo, per se.  Is the intent of this ticket to restructure the yaml files, launch files and config files within the CARMAPlatform repo?  Are all other repos already conformant?  Please add a little explanation.

msmcconnell, 2019-09-05:
Resolved by PR #323

closed 2019-09-05 by msmcconnell