usdot-fhwa-stol/carma-platform #321: Pacifica: Windshield wipers doesn’t work when Autoware/PACMOD is engaged during a Waypoint Following

maefromm, 2019-08-13:
### Types of Issue

- [x ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Creating the AS support ticket #2171 for the Pacifica issue we reported back in June 2019 during our dry run before a demo when it was raining. 

The windshield wipers not working when Autoware/PACMOD is engaged during a Waypoint Following dry run. Is this expected behavior or is this specific only to PACMOD or Autoware? 

This is a safety issue for us so need to know how to fix it.

TonysCousin, 2019-09-05:
@maefromm isn't there some additional info in an email or slack conversation? Would be good to have a copy of that here.

maefromm, 2019-10-07:
[RE AutonomouStuff Re RE EXTERNAL Request #2171 How would you rate the support you received.txt](https://github.com/usdot-fhwa-stol/CARMAPlatform/files/3697616/RE.AutonomouStuff.Re.RE.EXTERNAL.Request.2171.How.would.you.rate.the.support.you.received.txt)

maefromm, 2019-10-07:
@TonysCousin I have attached the AStuff ticket, and sent a followup.

maefromm, 2019-10-14:
Rinda Gunjala (AutonomouStuff) 
Oct 11, 11:12 AM CDT 

Mae,
We are currently experiencing some issues with our deployment pipelines so I just decided to go ahead and give you a built workspace with the bug fix.  I have attached a catkin workspace (called ssc_ws) along with this e-mail with the SSC binaries with the bug fix. Please extract it and source the setup.bash file in the install folder ($ echo "source <path to the where you extract the workspace on your computer>/ssc_ws/install/setup.bash --extend" >> ~/.bashrc && source ~/.bashrc) 

Please also remember to un-source the setup.bash file from your previous workspace with the SSC package that you are currently using before you try this one out.  We have tested this on our demo vehicle but let me know how it goes. I am sorry for the inconvenience- we are working on an easier way to roll out bug-fixes which should be ready shortly. 

Rinda Gunjala
AutonomouStuff, LLC
T: 309.291.0966 (Ext. 118) | F: 309.481.5425
rgunjala@AutonomouStuff.com | www.AutonomouStuff.com

maefromm, 2019-12-17:
Followed up via email to see it's been finally merged to their master and what version we should update to.