usdot-fhwa-stol/carma-platform #336: Fix CARMA 3.0 restart at the end of the route.

maefromm, 2019-08-29:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
After #300 is fixed, this restart issue at the end of the route should also be addressed especially for Demo purposes. 

CARMA2 has existing functionality to restart after the end of route, without having to shut down and restart. This is essential during demos when more than one trip is needed especially when it takes time to achieve localization. 

### Expected behavior
Be able to restart after end of route. 

### Actual behavior
Currently only re-engaging works in the middle of the route. 
At end of route, carma should be able to restart without shutting down.

### Related work

Link to related tickets or prior related work here. #300

kjrush, 2021-02-08:
After end of route behavior update, this is still an issue due to trajectory publication exception from Trajectory Executor after end of route message is published by Route node.

msmcconnell, 2021-03-22:
This functionality is working in 3.5

closed 2021-03-22 by msmcconnell