usdot-fhwa-stol/carma-platform #352: CARMA launch file does not have ability to pass map area list

qswawrq, 2019-09-13:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
In map_file package, ndt_matching.launch accepts map cell path as a ROS launch parameter. But top level carma launch file does not has this parameter placeholder.
### Expected behavior
Top level launch file should provide parameter placeholder to easily add new map cell paths.
### Actual behavior
Top level launch file does not provide parameter placeholder to easily add new map cell paths.
### Steps to reproduce the behavior
N/A
### Related work
N/A

qswawrq, 2019-09-13:
Current map loader does not need this rosparam.

closed 2019-09-13 by qswawrq

msmcconnell, 2019-09-13:
Functionality was added in usdot-fhwa-stol/autoware.ai#6