usdot-fhwa-stol/carma-platform #354: Rviz sometimes crashes when trying to open carma_defualt.rviz config file.

msmcconnell, 2019-09-13:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Launching Rviz using the carma_default.rviz file will sometimes result in a crash. It is not clear what is causing the issue as sometimes it will load without errors. Additionally, the majority of the configuration can be rebuilt from a default rviz config without an error occurring. 
### Expected behavior
Rviz launches reliably with desired configuration
### Actual behavior
Intermittent rviz crashes occur when opening carma config file. 
### Steps to reproduce the behavior

Launch the platform multiple times and try to open the carma_default.rviz file.

TonysCousin, 2019-09-13:
@msmcconnell Can you quantify the frequency of this occurrence?  What is the implication when it does happen - does the user need to shut down everything and relaunch Carma, or can you just try a second time to open the rviz file?

msmcconnell, 2019-09-16:
@TonysCousin The frequency is very inconsistent. Sometimes I simply just need to restart Rviz once or twice, but other times I cannot get the configuration to load at all and need to rebuild it manually from the default Rviz configuration. Restarting CARMA doesn't seem to have much impact on the behavior.