usdot-fhwa-stol/carma-platform #360: Update map loader to have a way to associate maps with transforms to ECEF

msmcconnell, 2019-09-16:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
In order to convert lat/lon into map frame a transform is needed from ECEF to map frame. This is currently loaded as a rosparam from a yaml file stored in the map_loader package. However, this makes is difficult to manage the argument with docker as it is not associated with the map files outside the docker image. There are a couple of ways to handle this.
1. Move the parameter file into /opt/carma/maps and have the map_loader node look for it there
2. Load it directly from the vector map
3. Add all the transforms into the parameter file as different parameters which are selected from at runtime (least extensible option). 
4. Some other option not listed

### Expected behavior
The user does not need to build a new docker image or create a new volume in order to change maps
### Actual behavior
The user currently needs to either change the parameter and rebuild the autoware image or create a volume pointing to a new yaml file on the host in order to switch map files.

msmcconnell, 2019-10-23:
Another option is to have the map load only after the route has been selected. Though this might make auto routing more difficult if we implement that in the future.

TonysCousin, 2020-04-17:
Closing per Mike McConnell.

closed 2020-04-17 by TonysCousin