usdot-fhwa-stol/carma-platform #369: Unit test build failure does not result in CI failure

msmcconnell, 2019-09-25:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
As seen in issue #368, CI will not trigger a failure even if the unit tests fail to build. Instead it will report no code coverage metrics available. It is likely this is caused by the sonarcloud build wrapper hiding the error code from making it to the top of the shell. A similar thing appears to be happening with the regular code build except it will cause a CI failure because SonarCloud analysis will throw an error code at the end of the CI step. 
### Expected behavior
Unit test build failure results in CI failure
### Actual behavior
Only regular build failures result in CI failure. Unit test build failures are reported as no code coverage metrics being available. 
### Steps to reproduce the behavior

PR with broken unit tests, but all other code builds.

TonysCousin, 2020-05-05:
Marking sev 4 because there is nothing functionally wrong with the production code, only how unit tests are being built.

jtbaird, 2020-05-07:
Resolved via carma-base-3.7.1 and autoware.ai-3.8.2 releases, as well as above PRs.

closed 2020-05-07 by jtbaird