usdot-fhwa-stol/carma-platform #370: CARMASscInterfaceWrapper docker image build failing do to dbw_mkz_msgs

msmcconnell, 2019-09-25:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The docker builds for the SSCInterfaceWrapper will fail because the SSC binaries were built against an older dbw_mkz_msgs package version. 
```
ssc_controller_driver      | [ERROR] [1569443442.721246260]: Client [/hardware_interface/veh_controller] wants topic /hardware_interface/ds_fusion/brake_report to have datatype/md5sum [dbw_mkz_msgs/BrakeReport/a92bad28c400885f36170c1cab44618e], but our version has [dbw_mkz_msgs/BrakeReport/d99bccbe9927a92147840ac1ddf11f87]. Dropping connection.
```

### Expected behavior
CARMASSCInterfaceWrapper repo can have it's docker images built and used on a vehicle.
### Actual behavior
Docker images do not build.

### Steps to reproduce the behavior

run the build_image.sh script in the CARMASscInterfaceWrapper repo

msmcconnell, 2019-10-09:
@qswawrq Can we close this issue? I believe you resolved it in usdot-fhwa-stol/CARMASscInterfaceWrapper#27

qswawrq, 2019-10-09:
> @qswawrq Can we close this issue? I believe you resolved it in [usdot-fhwa-stol/CARMASscInterfaceWrapper#27](https://github.com/usdot-fhwa-stol/CARMASscInterfaceWrapper/pull/27)

Sounds good to me.

closed 2019-10-09 by qswawrq