usdot-fhwa-stol/carma-platform #373: Skyline Release - Deployment and Configuration Fixes

maefromm, 2019-09-30:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
1. Update the autoware.ai mapfile param.yaml to use the latest map file. 
2. Update to use ndt_node only, instead of auto mode. 
3. Address any other configuration changes required to properly deploy the Skyline release to all the vehicles. 

### Related work
Ensure all the final code for this release have been checked in and tested.

TonysCousin, 2019-09-30:
@maefromm I changed the title from "fixes" to "changes" since a fix implies there is a defect. In fact, I see these as enhancements (since nothing is broken), so marked the ticket as such.

maefromm, 2019-10-23:
This can be closed with the Skyline being merged to master and the fix on 374.

closed 2019-10-23 by maefromm