usdot-fhwa-stol/carma-platform #381: Fusion CAN topic mismatch

qswawrq, 2019-10-04:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [] Enhancement request (describe the enhancement being requested)
- [] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Fusion cannot engage from CARMA3 because two topic name mismatch.
can_node publishes to ```/hardware_interface/can_bus_dbw/can_rx```, but dbw_node subscribes to ```/hardware_interface/ds_fusion/can_bus_dbw/can_rx``` 
Similar issue on ```/hardware_interface/can_bus_dbw/can_tx``` and ```/hardware_interface/ds_fusion/can_bus_dbw/can_tx```
### Expected behavior
Topic names are mapping correctly.
### Software version this applies to

Latest develop branch, commit 92828fa6b614abcf35b46a5ce3c69c7f53394e39
### Actual behavior
Two topic names are mapping incorrectly.
### Steps to reproduce the behavior
Run CARMA3 docker on Fusion.

### Related work
N/A

itabrl, 2019-10-09:
new update is fixing this issue.

closed 2019-10-09 by itabrl

maefromm, 2019-10-14:
@itabrl reminder to please link the Pull Request that closes the issue. I believe this is the one https://github.com/usdot-fhwa-stol/CARMASscInterfaceWrapper/pull/29 that resolved it.