usdot-fhwa-stol/carma-platform #383: Fix CARMA End of Route Issues - Braking Hard and Ending Sooner than Autoware

maefromm, 2019-10-09:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [] Enhancement request (describe the enhancement being requested)
- [] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During CARMA3 waypoint following, the vehicle slams on the brakes at the end of the route and ends much sooner compared to what Autoware does. 

### Expected behavior
Should be smooth deceleration at the end of route. 
Should end at the same spot Autoware does if CARMA is using the same purepursuit algorithm. 

### Software version this applies to

Develop Version as of 10/04/2019. 

### Actual behavior
Slams on the brakes compared to Autoware. Does not matter if the waypoints speed is smoothly decelerating to 0. CARMA3 just ends way before Autoware does and slams the brakes hard.

### Steps to reproduce the behavior

1. Run the CARMA3 waypoint following all the way to the end of the route.

### Related work
Another end of route issue that's documented on separate ticket is #300

maefromm, 2019-12-04:
Decided during Sprint 21 Planning that we are postponing this until Guidance and Route Manager completely handles the end of route.

kjrush, 2021-02-08:
CARMA end of route mechanism currently being tested in Vanden-plas doesn't exactly exhibit this behavior but may require tuning or adjustment to begin braking earlier at higher speeds to reduce jerk and acceleration.

msmcconnell, 2021-03-22:
Waypoint following is no longer used in 3.5.0 and this was also not an issue in 3.4.0

closed 2021-03-22 by msmcconnell