usdot-fhwa-stol/carma-platform #384: Poor localization performance

msmcconnell, 2019-10-09:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [] Enhancement request (describe the enhancement being requested)
- [] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
Using the new down-sampled demo map at TFHRC there are some strange localization behaviors. 
1. GPS based initialization no longer seems to be working on that map and the user must use rviz to define the starting point. Perhaps this is caused by a bad transform between the map and the ECEF frame. 
2. It seems NDT can find a lock even if visually the scan does not align. 
It is possible that the down-sampling may be behind this behavior, but that is just a theory. 
### Expected behavior
Localization works as well as the pre-release version of CARMA
### Software version this applies to

develop - 92828fa
### Actual behavior
Localization does not initialize properly 
NDT does not track as reliably. 
### Steps to reproduce the behavior

Run version 92828fa

msmcconnell, 2019-10-09:
Thoughts on cause
1. My guess is that the transform between the new map and the ECEF frame is not accurate.
2. Perhaps using a voxel size of 4 on an already down-sampled map is reducing NDT accuracy.

qswawrq, 2020-01-14:
Fixed by https://github.com/usdot-fhwa-stol/autoware.ai/pull/47

closed 2020-01-14 by qswawrq