usdot-fhwa-stol/carma-platform #402: build-image.sh has wrong path to get-system-version

Supraluminal, 2019-10-21:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
`docker/build-image.sh` references `./get-system-version.sh` at location in the script with the PWD is at the root of the repo instead of the `docker` folder. This affects all dockerized CARMA repos.

### Expected behavior
The `./get-system-version.sh` to be `./docker/get-system-version.sh`

### Actual behavior
```Tagged usdotfhwastol/carma-novatel-gps-driver:3.1.0 as usdotfhwastol/carma-novatel-gps-driver:latest
./build-image.sh: line 67: ./get-system-version.sh: No such file or directory
Error parsing reference: "usdotfhwastol/carma-novatel-gps-driver:" is not a valid repository/tag: invalid reference format
Tagged usdotfhwastol/carma-novatel-gps-driver:3.1.0 as usdotfhwastol/carma-novatel-gps-driver:```
### Steps to reproduce the behavior
Run `docker/build-image.sh`

### Related work

Link to related tickets or prior related work here.

maefromm, 2019-10-21:
Additional missmatch of the names usdotfhwastol/carmaavtvimbadriver to usdotfhwastol/carma-avt-vimba-driver

maefromm, 2019-10-23:
Additional mismatch of the names for Delphi esr and srr2 driver.

TonysCousin, 2020-04-21:
Per John Baird, this has been resolved.