usdot-fhwa-stol/carma-platform #404: Duplicate functionality in Autoware plugin and velocity_set node

qswawrq, 2019-10-22:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [x] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
In velocity_set node, it adds the current location as the first point in the final waypoint list. In the Autoware plugin ```AutowarePlugin::create_uneven_trajectory_from_waypoints```, it adds the current location to the trajectory point before it process Autoware final waypoints. Since those two are essentially the same functionality, the logic in Autoware plugin should be removed.
### Expected behavior
The adding initial waypoint logic should be only in the velocity_set.
### Software version this applies to

Skyline Release
### Actual behavior
The adding initial waypoint logic also exists in the autoware plugin.
### Steps to reproduce the behavior
N/A
### Related work
N/A

TonysCousin, 2020-04-17:
Shuwei & Mike both confirmed this is still an open issue.  The Autoware Plugin referred to is the in-lane cruising tactical plugin.

msmcconnell, 2021-03-01:
velocity_set node was removed in the vanden-plas release, so this issue can be closed.

closed 2021-03-01 by msmcconnell