usdot-fhwa-stol/carma-platform #406: Fix master branch for usdot-fhwa-stol/autoware.ai repository

maefromm, 2019-10-25:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During the skyline release, the Skyline release branch was merged by mistake into Master branch, instead of carma-master branch. Pull Request: https://github.com/usdot-fhwa-stol/autoware.ai/pull/28 

Per Kyle, using the GitHub Revert button will substantially make it more complicated to add the commits back to master in the future. Therefore need further discussion with John to fix this.  

### Expected behavior
PR should be reverted. 

### Software version this applies to

Skyline Release or 3.1.0

### Actual behavior
NA 

### Steps to reproduce the behavior
NA

### Related work

Link to related tickets or prior related work here.
NA

msmcconnell, 2019-11-07:
This issue is effectively OBE since Autoware 1.12 split the single repo into many the old master branch no longer applies to an active repository. Therefore it would probably be better for the updated CARMA fork to just start using the existing develop and master branches and deprecate the carma-develop and carma-master branches.

TonysCousin, 2020-01-14:
I have compared the master vs carma-master and they can be merged without conflict. Also, comparing develop vs carma-develop reveals several conflicts that will need to be resolved manually.  Since we are not using the develop branch, we can safely do this manual "merge" by selecting the carma-develop version of each of the several conflicting files.  In doing so, then we can go back to using both the conventionally named master and develop branches and deprecate the ones with the carma prefixes.

TonysCousin, 2020-01-14:
Further study reveals that the README.md at the top of the autoware.ai forked repo indicates that creating a merge that spans subtrees is a bad thing.  With these merges covering over 4000 files, it is inevitable that we will be spanning subtrees.  Therefore, I am choosing not to take that risk, or spend the time to figure out if/how we can get around it.  Rather, we can be okay living with our unconventional branch names, carma-master and carma-develop, and cease to use the regular branches in this repo.  As it stands, carma-master and carma-develop are in good shape for current use as they are - no repairs need to be done at this point.

closed 2020-01-14 by TonysCousin