usdot-fhwa-stol/carma-platform #420: Truck tablet is too big

TonysCousin, 2019-11-19:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During red truck acceptance testing at AStuff, the tablet supplied had two problems preventing its use as a Carma UI.  First, it was too large and heavy to safely mount anywhere. Second, it has an integral kickstand that gets in the way of most conventional mounting mechanisms.

### Expected behavior
Tablet can be safely mounted to the dashboard within easy vision range of the vehicle operator.

### Software version this applies to

n/a

### Actual behavior
Tablet has to be hand-held as it cannot be safely mounted anywhere in the cab that satisfies the operator needs.

### Steps to reproduce the behavior
n/a

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-01-21:
Ed Leslie has purchased the new tablet however mounting still an issue. Following up with Ed. 

The red truck has the new table in the box right now. The rest I believe will be delivered or has been delivered to AStuff and they're providing the mounting solution for the rest of the 3 trucks.

maefromm, 2020-01-23:
Ben/AStuff sent the link for the mount, leave issue open until this is order in place and tablet mounted. https://www.amazon.com/Industrial-Metal-Drill-Tablet-Mount/dp/B079TGWRNR/ref=sr_1_24_sspa?keywords=vehicle+mount+galaxy+tab+s5e&qid=1578946650&sr=8-24-spons&psc=1&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExVVVFUjhJSkhDSFZOJmVuY3J5cHRlZElkPUEwNzMzNTE4MU1MTTZKTUtGOEM3UCZlbmNyeXB0ZWRBZElkPUEwNDgzMTM0MzVZQzlMMDlMQlhQMCZ3aWRnZXROYW1lPXNwX2J0ZiZhY3Rpb249Y2xpY2tSZWRpcmVjdCZkb05vdExvZ0NsaWNrPXRydWU=

TonysCousin, 2020-04-17:
Tablets have been mounted in all trucks.  Issue resolved.

closed 2020-04-17 by TonysCousin