usdot-fhwa-stol/carma-platform #423: Update Autoware fork to accommodate truck configs

TonysCousin, 2019-11-19:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
As part of Autoware integration into the red truck, some config changes were made.  Ben at AStuff has promised to make a branch on their Autoware 1.12 fork (in Gitlab) to store these changes.

### Expected behavior
Leidos can pull changes directly from the AStuff fork.

### Software version this applies to

Autoware 1.12

### Actual behavior
Changes need to be manually added to the FHWA repo if this update isn't put in before the next truck is ready to test.

### Steps to reproduce the behavior

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-01-22:
Created ASTuff support ticket 2705 to provide updates to this ticket using this Github Issue.

TonysCousin, 2020-04-17:
Not sure if this is even relevant any more, since we have moved away from Autoware as a standalone installation.

msmcconnell, 2021-03-01:
I can't find the exact PRs, but this has been done and the trucks have been tested successfully since the Urus release

closed 2021-03-01 by msmcconnell