usdot-fhwa-stol/carma-platform #424: Gear mode fault on red truck

TonysCousin, 2019-11-19:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
After completing a successful run of Autoware at Rivian during Oct 2019, we tried to run again. However, the vehicle did not engage when requested. The SSC was reporting a Gear Module fault. The Pacmod lights did not turn red. However, cycling power to the pacmod resolved the issue. AStuff was provided with a ROS Bag of the problem.  After AStuff investigation, they determined this was the result of an autonomy sw (Autoware) error, and an erroneous error message appears during lack of autonomy input.  Fix is queued for next SSC release.

### Expected behavior
Pacmod should report this as a lack of autonomy input command.

### Software version this applies to

Pacmod 3

### Actual behavior
Pacmod reports this situation as a gear fault.

### Steps to reproduce the behavior
1. Start up the automation stack with Autoware
2. Cause Autoware to stop sending commands to the Pacmod

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-01-21:
Per Ben Ross on 01/17/2020, this was a case of slightly misleading diagnostics that we will target for future improvements. This was caused by restarting just Autoware and not restarting SSC.

maefromm, 2020-01-22:
Create AStuff support ticket 2710 and requested for updates using this Github Issue.

TonysCousin, 2020-04-17:
I believe AStuff fixed the misleading error message coming out of SSC in or before Feb 2020.  Need to verify that their ticket 2710 is closed.

maefromm, 2020-04-17:
@msmcconnell I think this was fixed on red truck, I assume when John tested it on the other trucks it was already included?

TonysCousin, 2020-07-27:
This problem was never observed on the silver, blue or white trucks during their acceptance in Feb 2020.  Mike says it wasn't seen on red either in Jan/Feb time frame.  So I am closing it.

closed 2020-07-27 by TonysCousin