usdot-fhwa-stol/carma-platform #426: Accommodate new gear shifting logic

TonysCousin, 2019-11-19:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [x] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
The trucks have different requirements on shifting from the cars. Notably they cannot shift out of Neutral (no parking gear) without the user pushing down the brake. It was found that Autoware did not request this behavior before trying to change gears and the SSC did not have extra logic to handle this condition. Therefore when Autoware attempted to engage the OEM Truck system would beep and prevent gear change. To get around this problem. TierIV and Leidos created a gear_change node which sat between the ssc_interface node and the SSC. This extra node provided the additional logic required. However, it makes more sense for this logic to actually reside in the SSC or ssc_interface. Leidos would prefer is reside in the SSC but AStuff was pushing back on that claiming the SSC's interface did not include gear shifting logic (though there was internal dispute of that).  

### Expected behavior
This is not a behavioral issue. The expectation is that the gear shifting delay logic will reside in AStuff code that we can pull from their Autoware fork (if in SSC_interface) or from a new release of SSC binaries.  Currently it does not.

### Software version this applies to

Autoware 1.12 AStuff fork.

### Actual behavior
Currently the behavior is coded into a separate (temporary) node that resides on the red truck.

### Steps to reproduce the behavior
See description above.

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-01-22:
As @TonysCousin mentioned, @saina-ramyar is currently addressing this issue and building this into CARMA SSC_interface_wrapper.

msmcconnell, 2020-01-29:
It turns out that Astuff has implemented this logic in the newest version of the SSC on the trucks so this change is no longer required. This issue can be closed.

TonysCousin, 2020-04-17:
Closed as OBE, per @msmcconnell comment above.

closed 2020-04-17 by TonysCousin