usdot-fhwa-stol/carma-platform #428: Rear-facing cameras are aimed too high

TonysCousin, 2019-11-19:
### Types of Issue

- [x] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [ ] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
During acceptance testing of the red truck at AStuff it was noted that the two rear-facing cameras on the cab and the one on top of the trailer were aimed too high.  The road surface only appears near the very bottom of the video frames.  Since the purpose of these cameras is for detection of nearby objects for collision avoidance, they need to see as much of the nearby pavement as possible. They need to be mounted with a more downward angle.

### Expected behavior
Can view vehicles or other objects near the truck.

### Software version this applies to

n/a

### Actual behavior
The only objects that can be seen on the roadway are at least 10 m away.

### Steps to reproduce the behavior

### Related work

Link to related tickets or prior related work here.

maefromm, 2020-01-15:
@TonysCousin Kyle and I saw it was pointed down but have not validated how it shows on RVIZ, should validate while it's at TFHRC. Not sure if Ben sent you pics to prove that it shows what you need or if you still want us to validate this.

maefromm, 2020-02-04:
@TonysCousin, during the 1/27 ATEF integration testing, @msmcconnell noticed that the cameras has an angled view after it was adjusted. https://drive.google.com/open?id=1wA69dv8HsUVZyfDvONAa0jd731-3LGsV. Need to discuss on how to fix this and what specs we should expect for the rest of the 3 trucks: 

1) How far down and how far out on the side should the camera view be? Currently the red truck will probably not see the vehicle in the next lane depending on how far it is. 
2) In addition, we need to figure out during the next acceptance testing of the the cameras: 
- overexposure 
- how to adjust it ourselves.

TonysCousin, 2020-04-17:
Calling this sev 4 now, since Carma doesn't have the capability to use the rear camera. But once that capability is in place, this should increase to sev 3 if still open.