usdot-fhwa-stol/carma-platform #431: Develop MVP OpenDrive to LaneLet2 Converter: Lane Geometry

itabrl, 2019-11-25:
### Types of Issue

- [ ] Anomaly report (something appears to not work correctly)
- [ ] Enhancement request (describe the enhancement being requested)
- [`x`] Other (please ensure the description clarifies why the issue doesn’t fall into either of the above categories)

### Descriptive summary
In order to use the LaneLet2 library for the world model as Autoware is in 1.13 we will need to convert our OpenDrive maps to LaneLet2's OSM format as the converter for Autoware is not yet built. This converter should be created under this task and need only cover lane geometry.

### Expected behavior
Validate the produced map by autoware_lanelet2_validation node

### Software version this applies to

Develop
### Actual behavior

### Steps to reproduce the behavior

1. Do this
1. Then do this...

### Related work

Link to related tickets or prior related work here.

msmcconnell, 2019-11-25:
Development of script began with #430 but more work is still needed

maefromm, 2019-12-17:
As discussed, @msmcconnell @TonysCousin we are closing this issue to include the merged code for the testarossa release. @msmcconnell please create a separate issue for the MVP portion.

closed 2020-05-05 by TonysCousin